ci-richard-mcelhinney/nhaystack #1: Signing jar using gradle

ChristianTremblay, 2020-09-29:
Previously, I was able to sign the module using gradle : 

```
// Include additional files in module jar 
jar { 
  from('src') { 
    include 'res/*.csv'
    }
    def passwd = "mypassworf"
    def app_alias = "my_alias"
    def jks_file = "C:\\0Programmes\\Java\\Eclipse\\my_keystore\\my_keystore"
    def tsa_url = "http://timestamp.digicert.com"
    def exec_line = "jarsigner -keystore " + jks_file + " -storepass " +
        passwd + " -keypass "+ passwd + " -tsa " + tsa_url + " " + jar.archivePath + " " + app_alias
    print  exec_line
    exec_line.execute()
}
```

Now I get `> Could not find method jar() for arguments` error.

I suspect gradle changed since I last used it.

I ended up using **jarsigner** directly on the command line. Could you share a way to include the signing "task" to gradle ?

Command line (for reference)
```
jarsigner -keystore "C:\\0Programmes\\Java\\Eclipse\\my_keystore\\my_keystore" -storepass mypassworf-keypass mypassworf -tsa http://timestamp.digicert.com C:\JCI\FXWorkbench-14.9\modules\nhaystack-rt.jar my_alias
```

```
jarsigner -keystore "C:\\0Programmes\\Java\\Eclipse\\my_keystore\\my_keystore" -storepass mypassworf-keypass mypassworf -tsa http://timestamp.digicert.com C:\JCI\FXWorkbench-14.9\modules\nhaystack-wb.jar my_alias
```

ci-richard-mcelhinney, 2020-10-23:
Hi Christian,

I have managed to get code signing working in our own modules and will shortly set it up for nhaystack as well.  There were some background issues that have now been cleared out of the way and I can move forward.

Cheers,
Richard

ChristianTremblay, 2020-10-24:
Fair enough. Won't need to self-sign future releases.
Still interested in knowing how you manage to sign from compile (without having to make the second manual step I had to make).

Closing this, thanks

closed 2020-10-24 by ChristianTremblay

ci-richard-mcelhinney, 2020-10-24:
No worries Christian.  As part of setting it up I'll be updating the gradle files to do the signing and I'll document how to do it.  Hopefully that will show you the steps for setting it up.

Cheers,
Richard

sjlongland, 2022-07-30:
Hi,

Silly question, I just stumbled on this very issue myself (having not touched Niagara development for a few years).  Niagara 4.11 has changed a lot since 4.4, but I did manage to produce some _unsigned_ `.jar` files -- which the workbench promptly rejected the moment I tried loading them.

Looking at the official `.jar`s, I see they're signed with a certificate that's in turn, signed by Symantec (according to `jarsigner -verify -certs -verbose <file>.jar`).  Thus it's the "standard" X.509 certificate signing procedure (rather than the Gradle signing plug-in, which uses OpenPGP).

I went looking through the various branches but did not spot any notes or steps relating to code signing in the current branches (I presume `n4-development` is still the "main" branch where the action happens).

Is this update still on the TO-DO list?

Thanks in advance.