ci-richard-mcelhinney/nhaystack #3: Connection keeps dropping out in 4.9

alperu, 2020-10-29:
WARNING [09:30:03 29-Oct-20 PDT][web] Invalid authorization header, scheme not supported: bearer
WARNING [09:35:03 29-Oct-20 PDT][nhaystack.watch] Watch ea19df5a-d851-4f55-8c1b-13beac44d8c6 timed out.

ci-richard-mcelhinney, 2020-10-30:
@esanderson can you advise if there have been any changes to the SCRAM authentiction process in 4.9?

ChristianTremblay, 2020-10-30:
Haven't had trouble with pyhaystack.

ci-richard-mcelhinney, 2020-10-30:
@ChristianTremblay are you using SCRAM as well?

esanderson, 2020-11-02:
@ci-richard-mcelhinney, I am not aware of any changes to SCRAM in 4.9.  I've reached out to the subject matter experts and will let you know if they come back with anything.

I tested the following combinations using digest authentication over TLS connections with both nhaystack 2.1.2 and 3.0.1.

r49 client to r49 server
r48 client to r49 server
r49 client to r48 server

By the way, is Auto Logoff Enabled set to false on the nhaystack server user with which you are connecting?

ChristianTremblay, 2020-11-03:
AFAIK we use SCRAM
https://github.com/ChristianTremblay/pyhaystack/blob/3c3508212c0f5a4e5f254fed4d169da91f03242d/pyhaystack/client/ops/vendor/niagara_scram.py#L20

I'm presently using 3.0.3 as I compiled and signed the actual repo to build our 4.9 release.

ci-richard-mcelhinney, 2020-11-09:
@alperu I assume you're using SkySpark as the client?  What version of SkySpark are you using?

ci-richard-mcelhinney, 2020-11-09:
I have done a lot of testing on this today.  Including building the nhaystack module under 4.9 for the first time and self-signing it.  I wrote an extra automated test in the Java test suite (not pushed up yet) that made a few thousand requests on 2 different threads and it never failed once.

So I changed direction and installed the latest SkySpark on my localhost and ran up a simple connection to a Niagara station also on my localhost.  My Niagara station only had 32 NumericWritable's in it and i configured SkySpark to do a simple poll over HTTP to the station and poll all of those points every second.

I also configured the Auto Logoff feature to be enabled with a timeout of 2 minutes which is the minimum I could configure in the Workbench.

By doing this I was able to replicate the issue and I observed it occurring for a period of an hour or so.  I finally managed to capture the interaction and noted that after a short period of time (just longer than 2 mintues) SkySpark tried to do a watchPoll and it was told the session id was unauthorised.  SkySpark subsequently immediately logged back in and started a new watch (it had a new watch id so that's how I knew it was a new watch) and continued on polling.

I then went and turned off the Auto Logoff feature for the user I was logging in as to do the watchPoll on the Haystack REST API and saw that the problem disappeared immediately.  In the Application Director of the station there were no more messages from the 'web' log or the 'nhaystack.watch' log indicating any error.  I also noted that in the Debug | Log screen of SkySpark, when enabling the logging for the haystack Ext, the transaction id for the Haystack watchPoll continued to count up, rather than be reset back to zero which was happening when Auto Logoff was enabled in Niagara.

At this stage, unless there is any further information, it seems to me that the remedy for this issue is to disable the Auto Logoff for the user that is logging into Niagara and doing the polling.

ci-richard-mcelhinney, 2020-11-11:
@alperu can you please confirm that what I said above resolves the issue so I can close this?  Otherwise I will need more information to do further investigation...

Cheers!

ci-richard-mcelhinney, 2020-11-16:
As there have been no further comments or information I am closing this ticket.  Please re-open if necessary.

Cheers,  
Richard

closed 2020-11-16 by ci-richard-mcelhinney

hvacsysint, 2020-11-18:
@ci-richard-mcelhinney I realize this issue has been closed, however I am trying to find a solution to establishing a connection with 4.9.  I've used the available version here and a signed version kindly provided by Alper, however I keep getting the same errors.  You mention above that you have established this connection, can you share specific and details on this?

Error 404 Not Found - using digest authentication
Login Failed - using httpbasic authentication

ci-richard-mcelhinney, 2020-11-18:
@hvacsysint if you would like to create a new issue and post some more details that would be appreciated so we can track it separately.

In the new issue, if you can post any errors from the Application Director of your Niagara installation that would be appreciated.