ci-richard-mcelhinney/nhaystack #4: Haystack-rt Module Usage Clarification

AlexH-BMS, 2020-11-09:
Hello,
Could you please confirm if a patched haystack module is still required when using Niagara 4.9?
Is dragging the Haystack tag dictionary from the N4Haystack module sufficient, or am I missing a step?

I'm having difficulty getting tags to be automatically applied, I've got as far as getting the haystack slot added if I do it manually, via the service. I can then add tags manually using the field editor.

ci-richard-mcelhinney, 2020-11-09:
Hi @AlexH-BMS , thanks for your question.

Yes, it is still necessary to add the Haystack Tag dictionary to the station.  I recommend to use the dictionary in Tridium's haystack-rt module.  I know this is a bit confusing and @esanderson and I are working to clarify this.

In 4.9, you can use the standard haystack-rt module, you don't need a patched haystack-rt module.  

@esanderson do you have any other comments on this?

esanderson, 2020-11-10:
There are 3 versions of the Haystack Tag Dictionary.

Tridium's haystack-rt module palette has one in a folder called "Standard items only" that once added to the Tag Dictionary Service displays "3.0.2 N.2" as the version.  This version has all of the Haystack tags, tag groups, and tag rules but does not contain any smart relations.  The dictionary in a folder called "Includes smart relations (recommended)" that displays "3.0.2 N.2 w/ SmartRefs (import)" as the version includes smart hs:equipRef and hs:siteRef relations.

A smart hs:equipRef relation is implied from any Control Point with a non-null Proxy Ext to the first ancestor component that has an hs:equip tag.  The smart hs:equipRef relation is not implied if there is a direct hs:equipRef relation already on the point.  If a Control Point with a non-null Proxy Ext has an hs:equipRef relation (either direct or implied) and that relation's endpoint (the "equip") has a direct hs:siteRef relation, a smart hs:siteRef relation is implied from the Control Point to the equip's hs:siteRef relation endpoint.

I am currently fixing a performance bug in 4.9 with the smart hs:siteRef relation that affects very large stations.  A patch for haystack-rt should be available soon.

The 3rd version of the Haystack Tag Dictionary is found in the version 3 beta nhaystack palette.  It displays "3.0.2 NH.2 (import)" as the version when added to Tag Dictionary Service.  It contains its own versions of the smart hs:equipRef and hs:siteRef relations that account for BHEquip components in the station.

There are 2 performance bugs affecting the smart hs:equipRef and hs:siteRef relations in the nhaystack module.  They are similar to the same bugs in Tridium's haystack-rt module and these bugs should be fixed soon.

AlexH-BMS, 2020-11-10:
Hi, thank you for the further clarification.
I've only been working with a supervisor station on my personal computer at present. I typically would use the haystack-rt (smart) dictionary. As the palette also contains your own it seemed to be implied that it should be used instead.
@esanderson Would you mind elaborating on the performance issues you mention?
Is this a runtime issue, or at discovery time when resolving the relations?
So if I'm following your reply correctly. The dictionary in the palette should be used if using the equip/site components, yet it is also possible to use the other dictionaries (implied or direct) to apply the relevant tags, rendering those components redundant?

ci-richard-mcelhinney, 2020-11-16:
@esanderson perhaps you could create a separate ticket here for the performance issues you mentioned then we can close this one as it sounds like @AlexH-BMS has had his question answered.

Cheers,
Richard

closed 2020-11-16 by ci-richard-mcelhinney