ci-richard-mcelhinney/nhaystack #10: hs:id tag is not applied to some custom components as an implied tag

ci-richard-mcelhinney, 2021-06-02:
I had some custom components in my module where I expected the hs:id tag to be applied.  However, when inspecting the implied tags on the component, and this was verified once the tags were shipped out of Niagara, the 'hs:id' tag wasn't there.  Looking at the rules in the tag dictionary it seems that the 'hs:id' tag should be applied to every baja:component.  

This needs to be looked into because the 'hs:id' tag is vital to a number of operations.

@esanderson you might need to check this out in the Tridium tag dictionaries.

esanderson, 2021-06-02:
The smart hs:id tag in the NHaystack module (BNIdTag) does an addition "visible" check (see SpaceManager#isVisibleComponent(BComponent)) in addition to the the tag rule conditions when determining whether or not to imply the tag.  This "visible" check includes the components that:

* The query user has read-access to
* Are instance of BHTagged (BHEquip and BHSite)
* Are instance of BControlPoint, BDevice, or BWeeklySchedule
* Have a slot named "haystack" whose type is instance of BHDict
* Have an "hs:site" or "hs:equip" tag (this check was added in more recent NHaystack versions)

If I recall correctly, this "visible" check was added to the smart tag to replicate how the NHaystackService formerly implied the ID tag.  The same "visible" check has always been used elsewhere in the NHaystack module.  In previous NHaystack versions, if a query was made for all components with the ID tag, it would only return components that met the "visible" conditions.

We could remove the extra "visible" check from the smart tag.  Then, the smart tag would be implied on all components for all NHaystack users.  If we want to allow users to choose whether the smart tag is implied on all components or only on "visible" ones, we could create a new smart tag type ("BNIdTag2") that removes the extra "visible" check.  Including this new type in the dictionary can be specified through a tag import file.  The Haystack tag dictionary is frozen to prevent changing the dictionary at runtime so a configuration property on the smart tag would not work.  The Haystack tag dictionary import does not support custom tag rule conditions, which might be another way to customize the smart tag's behavior.

ci-richard-mcelhinney, 2021-06-03:
@esanderson I've reviewed the code at SpaceManager#isVisible and I can see what you're referring to.  It all makes sense.  Rather than make too many changes would it simpler at here:

https://github.com/ci-richard-mcelhinney/nhaystack/blob/n4/nhaystack-rt/src/nhaystack/server/SpaceManager.java#L94

to just add an additional condition that includes any component that has the 'hs:point' tag applied to it.  This way we would capture the Site - Equip - Point hierarchy properly and the changes would be minimal.

So that line would turn into:
`if (tags.contains(NHaystackConst.ID_SITE) || tags.contains(NHaystackConst.ID_EQUIP) || tags.contains(NHaystackConstI.ID_POINT))
`

closed 2022-11-22 by ci-richard-mcelhinney