ci-richard-mcelhinney/nhaystack #13: History list request results, not displaying "dis" correctly

Tanner-1010, 2022-04-06:
Issuing the following request via postman returns a list of available histories.

https://xxx.xxx.xxx.xxx/haystack/nav?navId=his:/SITE00007_L001_O011

However the results returned display the dis and navName fields having identical values, when in this case the dis should show "ASHP Discharge Temperature"

Would it be possible to correct this if my assumptions are correct?

        {
            "tz": "London",
            "axHistoryId": "/SITE00007_L001_O011/P0006",
            "point": "m:",
            "dis": "SITE00007_L001_O011_P0006",
            "his": "m:",
            "axType": "s:history:HistoryConfig",
            "navName": "SITE00007_L001_O011_P0006",
            "id": "r:H.SITE00007_L001_O011.P0006 SITE00007_L001_O011_P0006",
            "kind": "Number",
            "unit": "°C"
        },

ci-richard-mcelhinney, 2022-04-08:
Hi @Tanner-1010 ,

there does seem to be some inconsistency as to how the "dis" tag is calculated in different scenarios.  Let me take a look and see if I can fix it up.

Tanner-1010, 2022-04-19:
Hi Richard,

Thanks for the prompt reply, do you have an ETA for the fix yet?

ci-richard-mcelhinney, 2022-04-28:
Hi @Tanner-1010,

Unfortunately doing this module isn't my regular day job! :) so I tend to do it in my own time at night.  I believe I know have the functionality working as you expect.

The next step is I need to do a release and for that I have had to renew my code signing certificate.  I hope to have this int he next day or two so fingers crossed I should be able to release something early next week.

Tanner-1010, 2022-04-29:
Hi Richard,
Thanks for your efforts to date, appreciate that you are working on this in your own time too.

Could you please comment on the following scenario too, would appreciate your insights.

We have several individual sites, each with a JACE 8000 installed and exporting their histories to a common Niagara v4.11 server.  We are querying these histories via the nHaystack connector with a view to importing them into another application for ML analytics, however no haystack tags are exported along with the history, even though they are present in the JACE and associated with the histories respective point.  We are having to manually re-add the haystack tags to the point at time of import and was wondering if you had any comment or can you identify any workaround?
I assume if not then we would need to contact Tridium to see if the associated points tags can some way be included along with the history at time of export.

Tanner-1010, 2022-07-26:
> Hi @Tanner-1010,
>
> Unfortunately doing this module isn't my regular day job! :) so I tend to do it in my own time at night. I believe I know have the functionality working as you expect.
>
> The next [...]

Hi Richard,

Just chasing for an update on the above please.

ci-richard-mcelhinney, 2022-11-22:
Hi @Tanner-1010 

So sorry for taking so long to get back to you.  I'd like to try to rectify this still even if you have worked around it.  I have one question, in the following statement:

> However the results returned display the dis and navName fields having identical values, when in this case the dis should show "ASHP Discharge Temperature"

Where does the display name "ASHP Discharge Temperature" come from?  Is it the display name on the point that has the history?  Is it a slot on the history extension?  Or something else?

Tanner-1010, 2022-11-22:
Hi Richard,

It is the display name on the point that has the history.  We would like this to be displayed as the display name of the history.
For this project we are exporting the histories from the JACE to a 2nd Niagara N4 server.  We then query this through the nHaystack connector to retrieve a list of available histories for a particular station.  It would be preferable if the display name of the history matched the display name of its associated point .
The excerpt I provided is a single result from a wider data set when I query the station "SITE00007_L001_O011" for all its associated stored histories.

closed 2023-04-17 by ci-richard-mcelhinney

ci-richard-mcelhinney, 2023-04-17:
I've merged in changes that should take care of this.  I will look to do a build shortly.