ci-richard-mcelhinney/nhaystack #15: Histories not appearing in HistorySpace

nassifo, 2022-06-28:
We found an issue when using this on N4 (we have N4.9 and latest Haystack driver) where histories that are brought up to the supervisor are not all appearing in the HistorySpace. This happens for points that were brought up to the supervisor and can be found in the ComponentSpace. We have tried to re-build the cache on the driver and still they do not appear. Any idea what could be causing this?

ci-richard-mcelhinney, 2022-06-28:
Hi @nassifo, 

I haven't had any other reports of an issue such as this.  I can try to do a quick test and see what happens.  Please give me a day to set this up and report back.

ci-richard-mcelhinney, 2022-06-30:
Hi @nassifo , 

I have setup a simple test today.  I setup 3 stations on my PC, 1 as a Supervisor and 2 simulating a Jace.  In each Jace station I had 4 NumericWritable points, each one had a Numeric COV history.  I linked a sinewave to each of the 4 Numeric Writables in each of the Jace stations.

I then connected the 2 Jace stations into the Supervisor using the Niagara Network and imported the 4 histories from each Jace station and I also added the 4 points from each Jace station into the Supervisor (under each Niagara Station device of course).

I was able to verify the histories were imported correctly and the points were being updated when they were subscribed.

I then added the NHaystackService and executed a query in my web browser, the query was `https://localhost/haystack/read?filter=point`.  This produced the following Zinc output.

`ver:"3.0"
cur,precision,curStatus,point,dis,curVal,axType,navName,id,axSlotPath,kind,axStatus,tz,his,axHistoryRef,axHistoryId,axPointRef
M,1.0,"ok",M,"Expiry",294,"nss:CertificateExpiryPoint","Expiry",@C.Services.SecurityService.certificates.tridium.expiry "Expiry","slot:/Services/SecurityService/certificates/tridium/expiry","Number","ok",,,,,
M,1.0,"ok",M,"NumericWritable",33.7744,"control:NumericPoint","NumericWritable",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable "NumericWritable","slot:/Drivers/NiagaraNetwork/Jace1/points/NumericWritable","Number","ok","Sydney",M,@H.Jace1.NumericWritable,,
M,1.0,"ok",M,"NumericWritable1",33.7744,"control:NumericPoint","NumericWritable1",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable1 "NumericWritable1","slot:/Drivers/NiagaraNetwork/Jace1/points/NumericWritable1","Number","ok","Sydney",M,@H.Jace1.NumericWritable1,,
M,1.0,"ok",M,"NumericWritable2",33.7744,"control:NumericPoint","NumericWritable2",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable2 "NumericWritable2","slot:/Drivers/NiagaraNetwork/Jace1/points/NumericWritable2","Number","ok","Sydney",M,@H.Jace1.NumericWritable2,,
M,1.0,"ok",M,"NumericWritable3",33.7744,"control:NumericPoint","NumericWritable3",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable3 "NumericWritable3","slot:/Drivers/NiagaraNetwork/Jace1/points/NumericWritable3","Number","ok","Sydney",M,@H.Jace1.NumericWritable3,,
M,1.0,"ok",M,"NumericWritable",63.2542,"control:NumericPoint","NumericWritable",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable "NumericWritable","slot:/Drivers/NiagaraNetwork/Jace2/points/NumericWritable","Number","ok","Sydney",M,@H.Jace2.NumericWritable,,
M,1.0,"ok",M,"NumericWritable1",63.2542,"control:NumericPoint","NumericWritable1",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable1 "NumericWritable1","slot:/Drivers/NiagaraNetwork/Jace2/points/NumericWritable1","Number","ok","Sydney",M,@H.Jace2.NumericWritable1,,
M,1.0,"ok",M,"NumericWritable2",63.2542,"control:NumericPoint","NumericWritable2",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable2 "NumericWritable2","slot:/Drivers/NiagaraNetwork/Jace2/points/NumericWritable2","Number","ok","Sydney",M,@H.Jace2.NumericWritable2,,
M,1.0,"ok",M,"NumericWritable3",63.2542,"control:NumericPoint","NumericWritable3",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable3 "NumericWritable3","slot:/Drivers/NiagaraNetwork/Jace2/points/NumericWritable3","Number","ok","Sydney",M,@H.Jace2.NumericWritable3,,
N,,,M,"NumericWritable",,"history:HistoryConfig","Jace1_NumericWritable",@H.Jace1.NumericWritable "NumericWritable",,"Number",,"Sydney",M,,"/Jace1/NumericWritable",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable
N,,,M,"NumericWritable1",,"history:HistoryConfig","Jace1_NumericWritable1",@H.Jace1.NumericWritable1 "NumericWritable1",,"Number",,"Sydney",M,,"/Jace1/NumericWritable1",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable1
N,,,M,"NumericWritable2",,"history:HistoryConfig","Jace1_NumericWritable2",@H.Jace1.NumericWritable2 "NumericWritable2",,"Number",,"Sydney",M,,"/Jace1/NumericWritable2",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable2
N,,,M,"NumericWritable3",,"history:HistoryConfig","Jace1_NumericWritable3",@H.Jace1.NumericWritable3 "NumericWritable3",,"Number",,"Sydney",M,,"/Jace1/NumericWritable3",@C.Drivers.NiagaraNetwork.Jace1.points.NumericWritable3
N,,,M,"NumericWritable",,"history:HistoryConfig","Jace2_NumericWritable",@H.Jace2.NumericWritable "NumericWritable",,"Number",,"Sydney",M,,"/Jace2/NumericWritable",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable
N,,,M,"NumericWritable1",,"history:HistoryConfig","Jace2_NumericWritable1",@H.Jace2.NumericWritable1 "NumericWritable1",,"Number",,"Sydney",M,,"/Jace2/NumericWritable1",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable1
N,,,M,"NumericWritable2",,"history:HistoryConfig","Jace2_NumericWritable2",@H.Jace2.NumericWritable2 "NumericWritable2",,"Number",,"Sydney",M,,"/Jace2/NumericWritable2",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable2
N,,,M,"NumericWritable3",,"history:HistoryConfig","Jace2_NumericWritable3",@H.Jace2.NumericWritable3 "NumericWritable3",,"Number",,"Sydney",M,,"/Jace2/NumericWritable3",@C.Drivers.NiagaraNetwork.Jace2.points.NumericWritable3
N,,,M,"Supervisor_AuditHistory",,"history:HistoryConfig","Supervisor_AuditHistory",@H.Supervisor.AuditHistory "Supervisor_AuditHistory",,"Str",,"Sydney",M,,"/Supervisor/AuditHistory",
[... cut ...]
As you can see if you read through the Zinc both the histories and the points are available over the REST API.

Be aware I did this test under 4.11 (we are not using 4.9 anymore as it is out of support from Tridium I think) and I was using the latest code, however the latest code is the same as the most recently published modules on Stackhub.org so there should be no difference.

I think I managed to capture your setup and design an accurate test but I was unable to replicate your issue.  Can you please provide more information so I can see if I need to do a different test.

nassifo, 2022-07-03:
Hi @ci-richard-mcelhinney 

Thanks for taking the time to set that up. I apologize for not going through this process myself to help narrowing down the issue. I got a chance this afternoon to go through a similar setup myself on Workbench 4.9 and I think I found our issue. 

I setup a normal Controller Station, a Jace Station, and a Supervisor Station on my Desktop. I tried to simulate our original setup as much as possible, we have points on the Controller Station which are brought up to the Jace Station where the trending extension is applied, and then both the trends and the points are further brought up to the Supervisor Station. Long story short, I was able to reproduce my issue and with both the `nav` op and the `read` op to my Haystack service, the histories were not appearing in the HistorySpace if the point is brought up under Points in the supervisor. 

Just to demonstrate, I have points "NumericWritable" and "BooleanWritable" with sine inputs on my Controller Station. I discover these points under my Jace Station, and I apply a NumericInterval and a BooleanCov on the Jace Station to these points. I also created a point originating on the Jace Station called "JaceNumeric" with a NumericInterval hist ext on it. Then, I bring these three points up to the Supervisor (All through NiagaraNetwork), as well as their respective histories. I run a Re-build Cache on the Haystack service and I type localhost:80/haystack/nav?navId="his:/JACE49" into my browser and this is what I see:

`ver:"3.0" dis:"javax.baja.sys.BajaRuntimeException: No nav histories found for 'JACE49'" along with a whole stacktrace`

Now, on the Supervisor under NiagaraNetwork/JACE49/points , I delete for example only the points (not trends), "BooleanWritable" and "JaceNumeric", Re-build Cache, and then I refresh my browser and now what I see is:

`ver:"3.0"
tz,axHistoryId,point,dis,his,axType,navName,id,kind,enum
"New_York","/JACE49/BooleanWritable",M,"JACE49_BooleanWritable",M,"history:HistoryConfig","JACE49_BooleanWritable",@H.JACE49.BooleanWritable "JACE49_BooleanWritable","Bool","false,true"
"New_York","/JACE49/JaceNumeric",M,"JACE49_JaceNumeric",M,"history:HistoryConfig","JACE49_JaceNumeric",@H.JACE49.JaceNumeric "JACE49_JaceNumeric","Number",`

I wasn't aware of this before but there is a setting in the Property Sheet of the NHaystackService called "Show Linked Histories" which I found to be set to "False" by default.  When I set this to "True" and then do a Re-build Cache, my histories show up in the HistorySpace even if the points are present under points. When I set it to "False" again, Re-build Cache, and then refresh my page, the histories disappear.

Now the odd thing is, I created another point on my Controller Station, put a trend on it on the Controller Station, brought the point and its trend from Controller to Jace, then from Jace to Supervisor. Now in every case I still see the trend and the point in both the HistorySpace and the ComponentSpace. Regardless of the value of "Show Linked Histories" and whether the point is brought up to the Supervisor. So the "issue", if you want to call it that, is only happening with points originating from a Jace Station it seems?

I haven't had time to test this out on 4.10/4.11 

Thanks for your help.

closed 2022-11-22 by ci-richard-mcelhinney