ci-richard-mcelhinney/nhaystack #17: Invalid tag name "ORHum"

christopher-hartley, 2022-08-25:
We have installed nhaystack verison v3.2.0.4.10.1.36 on our 4.10.0.154 instance of Niagara, and querying this Niagara instance from a 3.1.3 version of SkySpark. A contractor added two tags that are causing issues when we try to use haystackReadAll to look for records in the instance of Niagara from SkySpark (see below).

[image]

[image]

I've tried to run `readLink(@p:CLIENT:r:29a96504-7dd2889d).haystackRead(point)` [works], `readLink(@p:CLIENT:r:29a96504-7dd2889d).haystackReadAll(parseFilter(\\\"point and not ORHum\\\"))\")` [does not work], and `readLink(@p:CLIENT:r:29a96504-7dd2889d).haystackReadAll(equip)` [does not work], and `readLink(@p:CLIENT:r:29a96504-7dd2889d).haystackReadAll(parseFilter("point and fes!=ORHum and hs!=ORHum"))`. The returned error message is:

```
ver:"3.0" errType:"axon::EvalErr" err errTrace:"axon::EvalErr: Func failed: haystackReadAll(Expr conn,Expr filterExpr); args: (Call,Call)\n  haystack::CallErr: java.lang.IllegalArgumentException: Invalid tag name: ORHum [eval:1]\n=== Axon Trace ===\n  evalOrReadAll (eval:1)\n\n=== Remote Trace ===\njava.lang.IllegalArgumentException: Invalid tag name: ORHum\n  at org.projecthaystack.HDictBuilder.add(HDictBuilder.java:79)\n  at nhaystack.server.TagManager.generateComponentTags(TagManager.java:296)\n  at nhaystack.server.TagManager.createComponentTags(TagManager.java:463)\n  at nhaystack.server.SpaceManager\$CIterator.findNext(SpaceManager.java:230)\n  at nhaystack.server.SpaceManager\$CIterator.next(SpaceManager.java:211)\n  at nhaystack.server.SpaceManager\$CIterator.next(SpaceManager.java:190)\n  at nhaystack.collection.CompositeIterator.findNext(CompositeIterator.java:67)\n  at nhaystack.collection.CompositeIterator.next(CompositeIterator.java:39)\n  at org.projecthaystack.server.HServer.onReadAll(HServer.java:115)\n  at nhaystack.server.NHServer.onReadAll(NHServer.java:120)\n  at org.projecthaystack.HProj.readAll(HProj.java:141)\n  at org.projecthaystack.server.ReadOp.onService(HStdOps.java:149)\n  at org.projecthaystack.server.HOp.onService(HOp.java:48)\n  at org.projecthaystack.server.HServlet.onService(HServlet.java:90)\n  at org.projecthaystack.server.HServlet.doPost(HServlet.java:53)\n  at nhaystack.server.BNHaystackServlet.doPost(BNHaystackServlet.java:99)\n  at javax.baja.web.BWebServlet.service(BWebServlet.java:151)\n  at javax.baja.web.BWebServlet.doService(BWebServlet.java:195)\n  at javax.baja.web.BWebServlet.access\$000(BWebServlet.java:34)\n  at javax.baja.web.BWebServlet\$Servlet.doPost(BWebServlet.java:333)\n  at javax.servlet.http.HttpServlet.service(HttpServlet.java:707)\n  at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)\n  at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:763)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1655)\n  at com.tridium.web.filters.WebStartServletFilter.doFilter(WebStartServletFilter.java:47)\n  at org.eclipse.jetty.servlet.FilterHolder.doFilter(FilterHolder.java:193)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1642)\n  at com.tridium.web.filters.ContextFilter.doFilter(ContextFilter.java:42)\n  at org.eclipse.jetty.servlet.FilterHolder.doFilter(FilterHolder.java:193)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1642)\n  at com.tridium.web.filters.LocaleFilter.doFilter(LocaleFilter.java:31)\n  at org.eclipse.jetty.servlet.FilterHolder.doFilter(FilterHolder.java:193)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1642)\n  at com.tridium.web.filters.TridiumSecurityFilter.doFilter(TridiumSecurityFilter.java:42)\n  at org.eclipse.jetty.servlet.FilterHolder.doFilter(FilterHolder.java:193)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1642)\n  at com.tridium.web.filters.AddSubjectFilter.lambda\$null\$0(AddSubjectFilter.java:59)\n  at java.security.AccessController.doPrivileged(Native Method)\n  at javax.security.auth.Subject.doAs(Subject.java:422)\n  at com.tridium.web.filters.AddSubjectFilter.lambda\$doFilter\$1(AddSubjectFilter.java:58)\n  at java.security.AccessController.doPrivileged(Native Method)\n  at com.tridium.web.filters.AddSubjectFilter.doFilter(AddSubjectFilter.java:56)\n  at org.eclipse.jetty.servlet.FilterHolder.doFilter(FilterHolder.java:193)\n  at org.eclipse.jetty.servlet.ServletHandler\$CachedChain.doFilter(ServletHandler.java:1642)\n  at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:572)\n  at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)\n  at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:567)\n  at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)\n  at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:235)\n  at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1612)\n  at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:233)\n  at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1434)\n  at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:188)\n  at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:512)\n  at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1582)\n  at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:186)\n  at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1349)\n  at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)\n  at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:234)\n  at org.eclipse.jetty.server.handler.HandlerCollection.handle(HandlerCollection.java:146)\n  at com.tridium.jetty.NiagaraSecurityHandler.handle(NiagaraSecurityHandler.java:59)\n  at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)\n  at org.eclipse.jetty.server.Server.handle(Server.java:516)\n  at org.eclipse.jetty.server.HttpChannel.lambda\$handle\$1(HttpChannel.java:383)\n  at org.eclipse.jetty.server.HttpChannel.dispatch(HttpChannel.java:556)\n  at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:375)\n  at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:273)\n  at org.eclipse.jetty.io.AbstractConnection\$ReadCallback.succeeded(AbstractConnection.java:311)\n  at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:105)\n  at org.eclipse.jetty.io.ssl.SslConnection\$DecryptedEndPoint.onFillable(SslConnection.java:540)\n  at org.eclipse.jetty.io.ssl.SslConnection.onFillable(SslConnection.java:395)\n  at org.eclipse.jetty.io.ssl.SslConnection\$2.succeeded(SslConnection.java:161)\n  at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:105)\n  at org.eclipse.jetty.io.ChannelEndPoint\$1.run(ChannelEndPoint.java:104)\n  at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:336)\n  at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:313)\n  at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:171)\n  at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:129)\n  at org.eclipse.jetty.util.thread.ReservedThreadExecutor\$ReservedThread.run(ReservedThreadExecutor.java:375)\n  at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:773)\n  at org.eclipse.jetty.util.thread.QueuedThreadPool\$Runner.run(QueuedThreadPool.java:905)\n  at com.tridium.nre.util.PrivilegedRunnable.lambda\$run\$0(PrivilegedRunnable.java:30)\n  at java.security.AccessController.doPrivileged(Native Method)\n  at com.tridium.nre.util.PrivilegedRunnable.run(PrivilegedRunnable.java:28)\n  at java.lang.Thread.run(Thread.java:748)\n\n=== Fantom Trace ===\naxon::EvalErr: Func failed: haystackReadAll(Expr conn,Expr filterExpr); args: (Call,Call)\n  haystack::CallErr: java.lang.IllegalArgumentException: Invalid tag name: ORHum [eval:1]\n  axon::FantomFn.callx (FantomFn.fan:117)\n  axon::LazyFantomFn.callLazy (FantomFn.fan:161)\n  axon::Call.eval (Call.fan:39)\n  skyarcd::Context.evalOrReadAll (Context.fan:113)\n  axon::CoreLib.evalOrReadAll (CoreLib.fan:2547)\n  jdk.internal.reflect.GeneratedMethodAccessor256.invoke (Unknown)\n  jdk.internal.reflect.DelegatingMethodAc [... cut ...] ckReadAll(Expr conn,Expr filterExpr); args: (Call,Call)\n  haystack::CallErr: java.lang.IllegalArgumentException: Invalid tag name: ORHum [eval:1]"
```

My main questions are:
1. How can I query around these tags from inside SkySpark? What is the proper Axon query to exclude these points?
2. What about these tags are causing SkySpark to throw an error? Is the only way to fix this error to remove/edit these tags to be Axon-safe (https://skyfoundry.com/doc/lib-axon/func~toTagName)? 

Thank you!
4.

ci-richard-mcelhinney, 2022-08-26:
Hi @christopher-hartley, 

I've had a look at your question and I have something for you to try.  Firstly, we need to note that the actual exception is coming from the underlying haystack-java library written by SkyFoundry.  nhaystack uses this library extensively of course but nhaystack isn't generating the error.

Where the exception is being thrown it is doing a check for a valid tag name according to the Haystack rules for tag names.  The code that eventually does this check is [here](https://github.com/skyfoundry/haystack-java/blob/977612b7c584e5eeae2ca28331e66586c23bff3c/src/main/java/org/projecthaystack/HDict.java#L146)

If you follow this through it's checking to make sure that the first letter of the tag name is a lower case.  In your case the tag name is not lower case.  The rules for tag names are also documented [here](https://project-haystack.org/doc/docHaystack/Kinds#names)

The first thing I would like you to try is to change the tag names to be compliant with the tag naming rules.  From there we can examine if there are further issues.

christopher-hartley, 2022-09-15:
@ci-richard-mcelhinney following up on this issue, we had the integrators remove this tag for us from four different equips and four different points. Once removed, we were able to run a "haystackReadAll" from SkySpark without issue. I'd like to put forward a feature request that this error not be thrown (which prevents us from importing points into SkySpark) but either throws a warning and remove tags like this, or automatically force any uppercase tags to be haystack compliant (like the toTagName function in SkySpark). Let me know if you have more questions, and have a good week!

ci-richard-mcelhinney, 2022-09-15:
Hi @christopher-hartley , glad to hear we got to the bottom of the issue.  As it is resolved I will close this ticket.  In terms of your feature request it's noted, however this isn't something that is within the scope of nhaystack.  

Firstly, we would need to change the tag naming rules as documented on Project Haystack and then we would need to get the underlying haystack-java library which is actually throwing the issue.

Yes, I could wrap the error and swallow it and still add your tag anyway, but then nhaystack would be violating the naming rules for tags and I don't think that's a good idea.

I suggest you post on the Project Haystack forum and gauge what others think about this issue.  I've had no other feedback from the community that suggests this is an issue that is widely affecting the deployment of nhaystack.

closed 2022-09-15 by ci-richard-mcelhinney