ci-richard-mcelhinney/nhaystack #22: Cannot set duration

stvilla, 2023-01-03:
Hello,

I'm trying to use nHaystack to send a command to a point on a Jace (firmware version 4.8). I've managed to set a value on different levels, but I'm not being able to set a command duration. Here is the curl call that I'm doing to the server:

```
curl --location --request POST 'http://<server_ip_address>/haystack/pointWrite' \
--header 'Content-Type: text/zinc; charset=utf-8' \
--header 'Authorization: Basic <authentication>' \
--data-raw 'ver:"3.0"
id,level,val,who,duration
@S.site.GRUPPO1.fan_speed,8,1,"test",50ms
```

And this is the answer from the server:
```
ver:"3.0"
levelDis,level,val
"level 1",1.0,
"level 2",2,
"level 3",3,
"level 4",4,
"level 5",5,
"level 6",6,
"level 7",7,
"level 8",8,1.0
"level 9",9,
"level 10",10,
"level 11",11,
"level 12",12,
"level 13",13,
"level 14",14,
"level 15",15,
"level 16",16,
"level 17",17,
```
The duration just seems to be ignored. I've tried different unit of measurements and to send a raw number with no effect. 
Any idea on what I'm doing wrong?
Thank you

ci-richard-mcelhinney, 2023-01-09:
HI @stvilla , 

Thanks for your report.  I've just returned from summer vacation so please give me a few days to take a look and come back to you.

Regards,
Richard

stvilla, 2023-01-19:
Hi @ci-richard-mcelhinney,

do you have any update about the issue?

Thank you!

ci-richard-mcelhinney, 2023-04-17:
Hi, 

I can confirm that the duration is ignored.  A duration is used in Niagara when setting a timed override.  I will look to add functionality that allows this, however there is some risk.  If a permanent override is applied then who will release the override?

There is some functionality I need to explore here and check for edge cases.

ci-richard-mcelhinney, 2023-04-27:
Hi @stvilla and @eldritchhh, 

Thanks for your patience on this issue.

I have been investigating further and have concluded that the functionality already exists, however it is important to understand the priority array in Niagara.  The Niagara priority array for a point only allows timed overrides on level 8, so in8 for a Niagara Control point.  Just doing a point write and trying to set a duration for other levels won't work.

However, using the standard Haystack REST API Op 'invokeAction' you can override a point at level 8 for a timed duration and also release the override.  The action to override a point is the "override" action in Niagara and to release the override the "auto" action is invoked.  I have verified this behaviour with both the haystack-java library and also using cURL requests.  

The Java code using the haystack-java library is as follows:

```
    HDictBuilder args = new HDictBuilder();
    args.add("duration", 60, "sec");
    args.add("value", 100);
    HClient c = HClient.open("http://localhost/haystack", "user", "greatpassword");
    c.invokeAction(HRef.make("C.Apps.NumericWritable"), "override", args.toDict());
    Thread.sleep(10000);
    c.invokeAction(HRef.make("C.Apps.NumericWritable"), "auto", null);
```

The cURL commands I have executed are also shown below:

```
#!/usr/bin/env bash

curl --location --request POST 'http://localhost/haystack/invokeAction' \
-H "Content-Type: text/zinc; charset=utf-8" \
--user "basicuser:greatpassword" \
--verbose \
--data-raw 'ver:"3.0" action:"override" id:@C.Apps.NumericWritable
duration,value
60sec,100'

sleep 10s

curl --location --request POST 'http://localhost/haystack/invokeAction' \
-H "Content-Type: text/zinc; charset=utf-8" \
--user "basicuser:greatpassword" \
--verbose \
--data-raw 'ver:"3.0" action:"auto" id:@C.Apps.NumericWritable
empty
N'
```

On the basis that everything seems to be working as expected and it is possible to achieve a timed override using the standard REST API Ops, I'm going to close this ticket.  I will add some documentation around this to the Wiki in this Github repository as well so others can also understand how this works.

Thanks for reaching out!  Please let me know if I can assist further!

closed 2023-04-27 by ci-richard-mcelhinney