ci-richard-mcelhinney/nhaystack #27: Circuit Closed Exception when discovering points

MRSkeen, 2023-04-28:
Installed nhaystack-3.2.0-4.10.1.36 on Niagara_4_Developer-4.12.2.16.1.  Connected to remote SkySpark 3.1.5 server (see properties.png) and attempted to discover points.  AD log displays error (see ioexception.txt).  No problem discovering histories.  If you need more info, let me know what you need and I will provide it.  I haven't found any info online regarding this error related to nhaystack.  I cloned the driver and added some debug to BNHaystackLearnPointsJob.  Both of the for loops in run are completing.  325 key/value pairs are found and all are added, but the error is thrown before anything is added to the learn pane. 

[ioexception.txt](https://github.com/ci-richard-mcelhinney/nhaystack/files/11349652/ioexception.txt)
[image]
[image]
[image]

ci-richard-mcelhinney, 2023-05-01:
Hi @MRSkeen , I've been able to identify the issue.  I will send you a test module to see if my fix works via email.

MRSkeen, 2023-05-02:
Test module seems to be working fine.  I'm able to discover and add points now.

ci-richard-mcelhinney, 2023-05-05:
@MRSkeen thanks for the update.  I'll look at making a proper release now.

ci-richard-mcelhinney, 2023-06-04:
I am closing this as the issue is fixed and it is only pending a release.

closed 2023-06-04 by ci-richard-mcelhinney