ci-richard-mcelhinney/nhaystack #29: nHaystack enum string to Skyspark

taylornuttycombe, 2023-05-30:
Hi Richard,

Cross-posting this from the Skyspark forum post:

The problem seems to arise only from enum points. If we pull the information from the history extension on the Niagara server, the "Display" value is stored there so that is what we get; however, the curVal in Skyspark seems to be reading the "Tag" value (which does not support the problematic characters). See the link: https://www.dropbox.com/s/24ayqk2v6tbu2ad/String%20issue.bmp?dl=0

I am not sure what can be done about this, but this is the issue we are seeing.

Thanks,
Taylor

ci-richard-mcelhinney, 2023-05-31:
HI @taylornuttycombe , 

are you also able to provide some details of the configuration of the enum point on the niagara side please?

taylornuttycombe, 2023-06-01:
Sorry for the delay,

https://www.dropbox.com/s/86yykvx7ihgtmbv/PointDetails.bmp?dl=0

That is the property sheet for one of the enum points we are seeing the issue with.

closed 2023-08-14 by ci-richard-mcelhinney