ci-richard-mcelhinney/nhaystack #32: Schedule Event Invalid Zinc

FrancisMahony, 2023-06-27:
When attempting to use the pointWrite op to read a schedule event I get an issue parsing the zinc due to the ver:"3.0" line in the nested grid. Per a discussion on the SkyFoundry forum (https://skyfoundry.com/forum/topic/7109) it looks like it could be an encoding issue due to a missing "<<" and ">>".

ci-richard-mcelhinney, 2023-06-28:
Hi @FrancisMahony , 

Thanks for reporting.  This is actually an issue in the underlying haystack-java module that is maintained by Matthew Giannini.  I have raised this with him before and he is aware of this.  I am currently waiting on a fix in that module which will then in turn fix it in our module.

I will leave this ticket open for now as a reminder and follow up with Matthew to see where he is at with this one.

Cheers, 
Richard

ci-richard-mcelhinney, 2024-12-13:
@FrancisMahony this issue was fixed in the underlying haystack-java module and will be included int he upcoming release of nhaystack.  You can see the fix here: https://github.com/skyfoundry/haystack-java/commit/10af087928b3119463625604ac8bab9ba836bb67

closed 2024-12-13 by ci-richard-mcelhinney