ci-richard-mcelhinney/nhaystack #38: "Cannot load plugin" error

PaulRQuinn, 2024-02-19:
I'm using the nHaystack module 3.2.0-4.10.1.36 on a Niagara Supervisor 4.12.2.16.

When I add site and equip tags and then try to edit them in the Workbench, I get this error message:  "Cannot load plugin"

I have restarted the Supervisor.
 I have rebuilt the cache after adding sites.
I can make successful calls through the API and receive trends but I cannot filter for site.

Any suggestions on what might be causing this issue or troubleshooting steps?

Paul Quinn

ci-richard-mcelhinney, 2024-02-19:
Hi @PaulRQuinn , 

Thanks for reporting.  Let me load up the module and Niagara with those versions and take a look.  I'll come back to you shortly.

Cheers, 
Richard

ci-richard-mcelhinney, 2024-02-19:
Hi @PaulRQuinn, 

I've loaded up a station in 4.12.2.16 and the same nhaystack module off stackhub.org and for now I can't replicate the issue.  Could you possibly supply more details?

- How are you adding the 'site' and 'equip' tags in the first place?
- Which plugin are you trying to use to edit the tags?  The HTML Tag Manager view, or the Java popup dialog that allows you to edit the tags?
- Could you please provide the Application Director output of the Niagara Station?
- Could you please provide the console output of the Niagara Workbench?

Cheers, 
Richard

ci-richard-mcelhinney, 2024-12-13:
Closing this issue as there have been no recent updates.  Please re-open if more information can be provided.

closed 2024-12-13 by ci-richard-mcelhinney