ci-richard-mcelhinney/nhaystack #41: Authentication Flow

NeedleInAJayStack, 2024-03-24:
Hi, I'm working on a Haystack client library and have reports that when connecting to a Niagara 4 server with nHaystack installed, it is not getting the expected `401` response from a `HELLO` message to the `/about` op as documented in the [Project Haystack SCRAM Auth docs](https://project-haystack.org/doc/docHaystack/Auth#hello). Instead, it receives a `200`, but I'm not sure how to interpret that and what the ramifications are for re-authenticating later.

I tried to figure this out by looking at other implementations:
1. The Haxall haystack connector has [a short-circuit on `200` responses to the hello message](https://github.com/NeedleInAJayStack/haxall/blob/main/src/core/auth/fan/AuthClientContext.fan#L91)
2. pyHaystack appears to be using different endpoints for Niagara 4 - like hitting [`/prelogin`](https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/ops/vendor/niagara_scram.py#L136) and [`/j_security_check`](https://github.com/ChristianTremblay/pyhaystack/blob/master/pyhaystack/client/ops/vendor/niagara_scram.py#L174C20-L174C37) instead of `/about`

I have the following questions:
1. Does nHaystack follow the same Authentication flow and endpoints as the Project Haystack docs? If not, is there documentation on the authentication flows it uses?
2. If a client receives a `200: OK` from the `/about` endpoint, does that mean that the client may continue with subsequent haystack requests? Or does it mean that user has already authenticated and I may just use the provided cookie? Are there considerations for re-authenticating in this case?
3. What is the best way to run this package for live testing? Is a Niagara instance and license required?

ChristianTremblay, 2024-03-24:
I'm pretty sure the authentication comes from Niagara4 itself

NeedleInAJayStack, 2024-03-25:
Awesome, thanks for the quick and detailed responses guys. From your info and a deeper review of the `pyHaystack` and `haxall haystack` code, I feel like I've got a better handle on it. I'll close this issue out. Thanks again!

closed 2024-03-25 by NeedleInAJayStack