ci-richard-mcelhinney/nhaystack #42: Question: Release planned to include "Added checks for retrieving number facets in case of bad config"?

tblong, 2024-04-02:
Specifically interested if there is a release planned to include this commit: https://github.com/ci-richard-mcelhinney/nhaystack/commit/ec59e604115f03533657c6f1426d9e0a10f464d0

ci-richard-mcelhinney, 2024-04-12:
Hi @tblong , 

As an open source project the code is available for anyone to build.  Having said that it has been a while since I did a new build so I should do one soon.  I'll try to get one out.

The tricky part is testing it on different versions of Niagara.  So if you can help with that it would be appreciated.

tblong, 2024-04-12:
> You can use a self-signed cert or buy your own code signing certificate. All the certificate does is certify that whoever built the code stands by the code.

Yes, though it would not be an official build from this repo's cert and then the community would end up not benefitting.  I could fork then sign with our own cert, however that would just further fragment this work and cause confusion for end users over who's published version should be used.

Good to take this offline and help document any part of the manual testing steps taken.

tblong, 2024-05-23:
Hey @ci-richard-mcelhinney, still happy to help test here.  Let me know how you might like to take this offline.

ci-richard-mcelhinney, 2024-12-13:
For everyone else, @tblong and I worked together offline and I'm really pleased to advise @tblong has constructed a very useful test system for the nhaystack REST API testing.  I will upload this to the repository when I do the new release.

closed 2024-12-13 by ci-richard-mcelhinney