ci-richard-mcelhinney/nhaystack #43: Batch hisRead Supported?

ThomasMagnum308, 2024-04-09:
I'm running nhaystack on Niagara 4.11. Is the batch hisRead functionality supported?

Getting an UnknownNameException: range error.

POST /hisRead

ver:"3.0" range:"today"
id
@123.RmTmp
@123.RmRh

https://project-haystack.org/doc/docHaystack/Ops#hisRead

ThomasMagnum308, 2024-04-22:
A possible workaround I found would be to reference the axHistoryId and pull history from SQL.
Do you know why this value is only populated for a handful of points?

robhuntington, 2024-06-11:
Hi @ci-richard-mcelhinney  - have you been able to check if batch hisRead is supported.  I am getting the same range error as @ThomasMagnum308 

ver:"3.0" dis:"org.projecthaystack.UnknownNameException: range"

ThomasMagnum308, 2024-07-18:
Just to wrap this up. This functionality isn’t currently supported.

https://project-haystack.org/forum/topic/1122

closed 2024-12-13 by ci-richard-mcelhinney