ci-richard-mcelhinney/nhaystack #44: haystackInvokeAction for Enum point

csalazar90, 2024-07-15:
Hello, I am using the function haystackInvokeAction for Niagara Points that are not writable. I have been able to invoke actions "set" or "override" with this for Numeric Points, However, when I tried to do the same with Enumeric Point I get an error. the call I am using is: readLink(@p:devServer:r:2df512bd-e60e50c3).haystackInvokeAction(@C.Drivers.NiagaraNetwork.AHU01.points.B173~2d1.points.VavDmprOvrdCmd,"override",{value:"OPEN",duration:1min})

this Enum point has ranges for: OPEN, CLOSE, AUTO on Niagara.

the error I received is:

axon::EvalErr: Func failed: haystackInvokeAction(Obj conn,Obj id,Str action,Dict args); args: (GbRow,Ref,Str,Dict2) axon::EvalErr: Func failed: haystackInvokeAction(Obj conn,Obj id,Str action,Dict args); args: (GbRow,Ref,Str,Dict2)

haystack::CallErr: java.lang.ClassCastException: javax.baja.sys.BString cannot be cast to javax.baja.sys.BDynamicEnum [eval:1]
Can you help me to find a solution?

Thanks,

Carlos

naimakov, 2024-07-16:
Hi Carlos,

I have a solution ready for this. I can raise a PR if someone from contributors or @ci-richard-mcelhinney would allow me to branch out from `n4` branch.

Best regards,
Nurkanat

ci-richard-mcelhinney, 2024-07-16:
Hi @csalazar90, 

thanks for reporting.  Are you able to provide more information from the Niagara side?  I'd like to get a clear picture of the error that's occurring in nhaystack itself.  You've provided the output from SkySpark which doesn't have all the details.  Can you upload a stacktrace from the Niagara Application Director please?

@naimakov I'm happy to review your patch.  You don't need to branch from this repo.  You can fork this repo and apply your patch then create a PR from your forked repo to this one.

csalazar90, 2024-07-16:
Hi @ci-richard-mcelhinney thanks for replying. I attached stacktrace from both sides on N4 and SkySpark. please let me know if you need more information.
Thanks
[tractrace from skyspark.txt](https://github.com/user-attachments/files/16250292/tractrace.from.skyspark.txt)
[stacktrace on N4.txt](https://github.com/user-attachments/files/16250293/stacktrace.on.N4.txt)

ci-richard-mcelhinney, 2024-07-23:
@naimakov did you have a PR you were able to share via a forked repo?  Or do you want to send me a diff?

naimakov, 2024-07-23:
@ci-richard-mcelhinney Hi, sorry, yes, I can share via a forked repo. I'll let you know once I push a PR.

naimakov, 2024-07-29:
@ci-richard-mcelhinney I just opened a PR for this issue #45

ci-richard-mcelhinney, 2024-12-13:
I am closing this issue as it will be resolved in the upcoming release of nhaystack.

closed 2024-12-13 by ci-richard-mcelhinney