ci-richard-mcelhinney/nhaystack #46: nhaystack-3.2.0 with Niagara 4.12?

fishe333, 2024-08-05:
I have a project with a soft jace at 4.12.2.16, the nhaystack-3.2.0-4.10.1.36 modules are giving me a signature error. Should these work with version 4.12 or am I doing something wrong?

ci-richard-mcelhinney, 2024-08-06:
HI @fishe333 , 

Thanks for reporting.  Can you provide some more information please?

- Where did you get these modules from?  Stackhub?
- Can you show the output of the Application Director for the Station or the Workbench console?
- In the Workbench can you go to the Tools -> Module Info menu item, click on it and find the nhaystack-rt module in the list, and double click on it and please post the screenshot that has the details of the module?

I will also try to find time to check it out this week.

fishe333, 2024-08-06:
Good morning Richard. Thanks for the quick reply. Yes, the module was downloaded from StackHub. I am actually a little confused about something, the module info shows an older version than Software Manager. I had copied the upgraded version (3.2.0-4.10.1.36) into the sw/inbox and restarted the Workbench. The Software Manager shows the newer version and the Module Info shows the older version (3.0.1). Did I upgrade this improperly? Where does Module Info get its info from as opposed to Software Manager? While atempting to resolve the issue I copied the newly unzipped nhaystack files directly into the modules folder but, no change. Below are srceen captures of both the Module Info details and the Software Manager details. Thanks!

[BC Application Director Output.txt](https://github.com/user-attachments/files/16509704/BC.Application.Director.Output.txt)

[image]
[image]

ci-richard-mcelhinney, 2024-08-06:
Thanks @fishe333 for providing that extra info.  

I agree this is very confusing information.

The next step I would take is to inspect inside the nhaystack-rt.jar file in your <niagara_home>\modules directory.  A jar is just a Zip file so you can open it with any zip tool.  Take a look at the module.xml file inside the nhaystack-rt.jar file in the 'modules' directory and check the version there.

You could also temporarily remove the nhaystack modules from the 'sw\3.2.0\' folder and repeat and see what results you get.

If my memory is correct, the 'sw' folder just keeps previous versions of modules and dist files if you need to downgrade.

closed 2024-09-05 by ci-richard-mcelhinney