ci-richard-mcelhinney/nhaystack #47: NHaystack - Not Running, No output

robshep, 2024-10-02:
Hi,

Apologies if this is too much of a support request rather than a software issue....

I've loaded both Jars into the workbench modules folder and I can see the module is loaded. 
But there is no output on system log concerning the module and no NHaystackService as the readme mentions. 

I've seen in other comments a log file entry that shows the module is starting and the cache is being built, but no output here.

I'm new to Niagara but plenty of experience in these kinds of applications and environments.

How can I increase some logging or tracing to see if there any errors.

System: 4.12.2.16

The module is thus:

[image]

Thank you for any pointers - it'd be great to get this going.

ParityDan, 2024-11-26:
@robshep - are you certain you are loading the *nhaystack* jar? your screenshot above is for haystack.

you will have to install the nhaystack service under the services container in your station to get the debug messages.

ci-richard-mcelhinney, 2024-12-13:
@robshep as @ParityDan has advised it doesn't look like you are loading the nhaystack jar file.  Try installing the nhaystack-rt.jar and nhaystack-wb.jar files and then install the NHaystackService from the palette into your station and see how you go.

closed 2024-12-13 by ci-richard-mcelhinney