ci-richard-mcelhinney/nhaystack #48: Cannot parse HRef'null' in haystack property of equip

jjlaber-dev, 2024-12-19:
Running nHaystack 3.3.0.0 on N4.14.0.162 I am getting a null pointer error when clicking on the arrow next to the haystack property on an HEquip object. I tested on 2 different stations of the same N4 version with equal results.

[image]

**Std Out:**

_javax.baja.sys.BajaRuntimeException: Could not parse HRef 'null'.
 at nhaystack.NHRef.make(NHRef.java:30)
 at nhaystack.server.TagManager.doLookupComponent(TagManager.java:149)
 at nhaystack.server.TagManager.lookupComponent(TagManager.java:144)
 at nhaystack.server.NHServer.onReadById(NHServer.java:181)
 at org.projecthaystack.HProj.readById(HProj.java:57)
 at org.projecthaystack.HProj.readById(HProj.java:47)
 at nhaystack.server.BNHaystackService.doReadById(BNHaystackService.java:925)
 at auto.nhaystack_server_BNHaystackService.invoke(AutoGenerated)
 at com.tridium.sys.schema.ComponentSlotMap.invoke(ComponentSlotMap.java:1921)
 at com.tridium.sys.schema.ComponentSlotMap.invoke(ComponentSlotMap.java:1884)
 at javax.baja.sys.BComponent.invoke(BComponent.java:1232)
 at com.tridium.fox.sys.broker.BBrokerChannel.invoke(BBrokerChannel.java:1895)
 at com.tridium.fox.sys.broker.BBrokerChannel.process(BBrokerChannel.java:248)
 at com.tridium.fox.sys.BFoxConnection.process(BFoxConnection.java:470)
 at com.tridium.fox.session.SessionDispatcher.dispatch(SessionDispatcher.java:92)
 at com.tridium.fox.session.SessionDispatcher.run(SessionDispatcher.java:68)
 at java.lang.Thread.run(Thread.java:750)_

ci-richard-mcelhinney, 2024-12-20:
Hi @jjlaber-dev, thank you for your report.

Support for the 'haystack' slot on components is mostly deprecated now, however I will try to find time to investigate this issue.

ci-richard-mcelhinney, 2024-12-20:
@jjlaber-dev could you please advise the steps you took to see this issue so I can accurately replicate it?

ci-richard-mcelhinney, 2026-09-22:
I believe this issue is solved.  The code is available on the `n4`branch.  I will try to do a release soon.  If you feel this should be re-opened please do so.

closed 2026-09-22 by ci-richard-mcelhinney