ci-richard-mcelhinney/nhaystack #49: NHWatch.sub cannot subscribe to S.site

BMST3ch, 2025-01-23:
I'm using the nHaystack module 3.2.0-4.10.1.36 on a Niagara JACE 4.10.10.40

From the Application Director, I'm receiving multiple messages like this one:

WARNING [07:08:37 23-Jan-25 PST][nhaystack.watch] NHWatch.sub 70c8a3f6-bb2a-42ed-8aaa-b29b2736b3c6 cannot subscribe to S.site.FourthFloorVavs-V405.ReheatPos

I have the NHaystackService "Enable"
I have the Haystack Tag Dictionary

How can I resolve this error?

Kind regards,

ci-richard-mcelhinney, 2025-02-04:
HI @BMST3ch ,

What type of Niagara component is the 'ReheatPos' component?

BMST3ch, 2025-02-05:
It is a proxy point from a Distech Controls ECB-VAV controller. We found that the issue was caused by the point being renamed.

Thanks for replying.

closed 2025-04-09 by ci-richard-mcelhinney