ci-richard-mcelhinney/nhaystack #54: Does Niagara 4.15 not support it?

imic, 2025-06-13:
javax.baja.sys.BajaRuntimeException: Could not parse HRef 'null'.
   at nhaystack.NHRef.make(NHRef.java:30)
   at nhaystack.ui.Row.makeRefFE(Row.java:301)
   at nhaystack.ui.Row.makeValueFE(Row.java:214)
   at nhaystack.ui.Row.<init>(Row.java:73)
   at nhaystack.ui.Row.<init>(Row.java:53)
   at nhaystack.ui.BHDictEditor.loadMainGrid(BHDictEditor.java:240)
   at nhaystack.ui.BHDictEditor.<init>(BHDictEditor.java:207)
   at nhaystack.ui.BHDictEditorGroup.<init>(BHDictEditorGroup.java:215)
   at nhaystack.ui.BHDictFE.edit(BHDictFE.java:136)
   at nhaystack.ui.BHDictFE.access$200(BHDictFE.java:55)
   at nhaystack.ui.BHDictFE$Popup.doInvoke(BHDictFE.java:131)
   at javax.baja.ui.Command.doInvoke(Command.java:313)
   at javax.baja.ui.Command.invoke(Command.java:283)
   at javax.baja.ui.BButton.doInvokeAction(BButton.java:151)
   at javax.baja.ui.BAbstractButton.mouseReleased(BAbstractButton.java:552)
   at javax.baja.ui.BWidget.fireMouseEvent(BWidget.java:1369)
   at com.tridium.ui.awt.MouseManager.fire(MouseManager.java:392)
   at com.tridium.ui.awt.MouseManager.fire(MouseManager.java:305)
   at com.tridium.ui.awt.MouseManager.released(MouseManager.java:136)

ci-richard-mcelhinney, 2025-06-16:
@imic I haven't had a chance to do any testing on 4.15 yet as it's only just been released.

Can you please provide more information on how to replicate the issue?

jgerster-wetec, 2025-07-17:
Same error here with Niagara 4.14.0.162.
The problem is also reported in https://github.com/ci-richard-mcelhinney/nhaystack/issues/48.
The issue occurs with a fresh station for testing purposes. After enabling NHaystackService, adding a site and an equip component, when I try to reference the site in the equip, the error occurs. I rebuild cache as mentioned in the docu.

[image]

dmauro-ieti, 2025-08-20:
Hello,
i have the same issue on 4.14.0.162:
- the site component seems to work correct, it appears after rebuild cache action and can be edited
- the equip component instead result the issue on the attach
[image]

ci-richard-mcelhinney, 2025-09-18:
@imic can you please provide some extra information as I look into this?

1. Which Haystack Tag Dictionary were you using in Niagara?  The Haystack 3 dictionary or the Haystack 4 dictionary?  
2. Did you have a siteRef relation going from the equip component to the site?

dmauro-ieti, 2025-09-18:
Hello Richard,
So in my test i was using dictionary version 4 and no, i didn'have the
relation between components

W1Z2Q3, 2025-11-11:
You should manually add relation(hs:siteRef) to equip where you want to open "Haystack"

[image]

ci-richard-mcelhinney, 2026-09-22:
I believe this is now fixed and the code is available on the 'n4' branch.  I will be trying to do a build soon.  If anyone feels the issue needs to be re-opened, please do so.

closed 2026-09-22 by ci-richard-mcelhinney